Nuclei-Software/nuclei-sdk #7: openocd reg command

quantrpeter, 2020-09-07:
Hi
   openocd reg command can't display all values of all registers. So i have to invoke command "reg a0", "reg a1" and so on, which is very slow. Any workaround?
thanks
Peter

fanghuaqi, 2020-09-07:
Hi, you can run `info reg all-registers`, but it will not display the customized csrs, for customized csrs, you still need to use `info reg csrxxx`, `csrxxx` replace to the real csr number such as `csr775` for mtvt

fanghuaqi, 2020-09-07:
You can also take a try with Nuclei Studio which has a gui debugger interface, see here https://www.nucleisys.com/download.php

quantrpeter, 2020-09-07:
Seems not working, i am using riscv-none-embed-gdb, installed by xpm. What gdb i have to use?

[image]

fanghuaqi, 2020-09-07:
Sorry, my mistake, it should be `info reg all`. 

[image]

quantrpeter, 2020-09-14:
Thanks, it works. But it is very slow. Is there any way to get all registers value by openocd please?

fanghuaqi, 2020-09-14:
Sorry, I didn't find a way to read all registers by openocd.

quantrpeter, 2020-09-14:
Thanks anyway

closed 2020-09-14 by quantrpeter