Nuclei-Software/nuclei-sdk #10: Batch execution raised an error

advancedwebdeveloper, 2020-11-11:
>setup.bat

> ****               ПРЕВЫШЕНИЕ  РАЗМЕРОВ  СТЕКА                ****
> ****      ПРИ  ВЫПОЛНЕНИИ  РЕКУРСИИ  В  ПАКЕТНОМ  РЕЖИМЕ      ****
> Глубина рекурсии=317, Использование стека=90 проц.
> ****              ПАКЕТНАЯ   ОБРАБОТКА  ПРЕРВАНА              ****

advancedwebdeveloper, 2020-11-11:
Hi.
I got this on Windows 10.

advancedwebdeveloper, 2020-11-11:
I would suggest that you might move to build automation scripts in Python, Powershell or use pre-compiled tools (in Golang).
cmd is a bad option, really.

fanghuaqi, 2020-11-11:
Hi @advancedwebdeveloper , thank you for your suggestion, we use `cmd` because we want user can execute in windows without install other bash tool.

advancedwebdeveloper, 2020-11-12:
https://please.build/ and https://ninja-build.org could be of help

fanghuaqi, 2020-11-12:
ok, thank you for this information of the build tool, I will close this issue.

closed 2020-11-12 by fanghuaqi

fanghuaqi, 2020-11-12:
> 
> 
> > setup.bat
> 
> > ****               ПРЕВЫШЕНИЕ  РАЗМЕРОВ  СТЕКА                ****
> > ****      ПРИ  ВЫПОЛНЕНИИ  РЕКУРСИИ  В  ПАКЕТНОМ  РЕЖИМЕ      ****
> > Глубина рекурсии=317, Использование стека=90 проц.
> > ****              ПАКЕТНАЯ   ОБРАБОТКА  ПРЕРВАНА              ****
> 
> Issue translated using google:
> 
> > setup.bat
> 
> > **** OVERSTACK SIZES ****
> > **** WHEN PERFORMING RECURSION IN BATCH MODE ****
> > Recursion depth = 317, Stack usage = 90 percent.
> > **** BATCH PROCESSING INTERRUPTED ****

For this issue, you need to prepare the `setup_config.bat` file  in Nuclei SDK root first, and then open windows cmd terminal in the Nuclei SDK Root, these steps are for windows.

1. Create`setup_config.bat` file.
   
   Assume you have extracted [Nuclei Studio IDE](https://www.nucleisys.com/download.php) to `D:\Software\NucleiStudio_IDE_202009\`.

   ~~~
   set NUCLEI_TOOL_ROOT=D:\Software\NucleiStudio_IDE_202009\NucleiStudio\toolchain
   ~~~

2. Then you can run `setup.bat` only in Nuclei SDK root to setup build environment.

    ~~~
    D:\workspace\Sourcecode\nuclei-sdk>type setup_config.bat
    set NUCLEI_TOOL_ROOT=D:\Software\NucleiStudio_IDE_202009\NucleiStudio\toolchain

    D:\workspace\Sourcecode\nuclei-sdk>setup.bat
    Setup Nuclei SDK Tool Environment
    NUCLEI_TOOL_ROOT=D:\Software\NucleiStudio_IDE_202009\NucleiStudio\toolchain

    D:\workspace\Sourcecode\nuclei-sdk>cd application\baremetal\helloworld
   D:\workspace\Sourcecode\nuclei-sdk\application\baremetal\helloworld>make all
   Current Configuration: RISCV_ARCH=rv32imafdc RISCV_ABI=ilp32d SOC=hbird BOARD=hbird_eval CORE=n307fd DOWNLOAD=ilm
   "Assembling : " ../../../SoC/hbird/Common/Source/GCC/intexc_hbird.S
   "Assembling : " ../../../SoC/hbird/Common/Source/GCC/startup_hbird.S
   "Compiling  : " ../../../SoC/hbird/Common/Source/Drivers/hbird_gpio.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Drivers/hbird_uart.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/clock_getres.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/clock_gettime.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/clock_settime.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/close.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/execve.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/exit.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/fork.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/fstat.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/getpid.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/gettimeofday.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/isatty.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/kill.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/link.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/lseek.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/open.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/read.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/sbrk.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/stat.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/times.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/unlink.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/wait.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/Stubs/write.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/hbird_common.c
   "Compiling  : " ../../../SoC/hbird/Common/Source/system_hbird.c
   "Compiling  : " main.c
   "Linking    : " helloworld.elf
      text    data     bss     dec     hex filename
      7826     112    2290   10228    27f4 helloworld.elf
    ~~~

Thanks
Huaqi