Nuclei-Software/nuclei-sdk #16: How to handle different variations of Longan Nano?

tuupola, 2020-11-28:
Currently there is two known versions of Longan Nano. The [official version](http://longan.sipeed.com/en/) has GD32VF103CBT6 which has 128K flash and 32K sram. Apparently some Aliexpress sellers also make version with GD32VF103C8T6 which has 64K flash and 20K sram. This is also sometimes referred as the Longan Nano Lite.

Currently the Nuclei SDK linker script for Longan Nano assumes the [Lite version](https://github.com/Nuclei-Software/nuclei-sdk/blob/master/SoC/gd32vf103/Board/gd32vf103c_longan_nano/Source/GCC/gcc_gd32vf103_flashxip.ld#L53). This causes the linking to fail with memory heavy applications.

```
Linking    :  firmware.elf
/opt/nuclei/gcc/bin/../lib/gcc/riscv-nuclei-elf/9.2.0/../../../../riscv-nuclei-elf/bin/ld: firmware.elf section `.bss' will not fit in region `ram'
/opt/nuclei/gcc/bin/../lib/gcc/riscv-nuclei-elf/9.2.0/../../../../riscv-nuclei-elf/bin/ld: section .stack VMA [0000000020004800,0000000020004fff] overlaps section .bss VMA [0000000020000268,0000000020006a23]
/opt/nuclei/gcc/bin/../lib/gcc/riscv-nuclei-elf/9.2.0/../../../../riscv-nuclei-elf/bin/ld: region `ram' overflowed by 0 bytes
collect2: error: ld returned 1 exit status
../nuclei-sdk-roman/Build/Makefile.rules:71: recipe for target 'firmware.elf' failed
make: *** [firmware.elf] Error 1
```

When changing the linker script to support 32kB sram linking succeeds.

```
Linking    :  firmware.elf
   text	   data	    bss	    dec	    hex	filename
  72022	    612	  28604	 101238	  18b76	firmware.elf
```

This issue is to start discussion how to support the different variations of the Longan Nano board. Separate linker scripts chosen with a Makefile variable is one possibility.

fanghuaqi, 2020-11-30:
I would like to provide variable called `VARIANT`, in Makefile.SoC.xxx to handle the small differences.

If `VARIANT := lite`, it will use another linker script for this board.

RomanBuchert, 2020-11-30:
Is it also possible to set the SOC to gd32vf103 by default for the longan nano board?

tuupola, 2020-11-30:
@fanghuaqi Ok. Do you want me to take shot at this? Would be a good way for me to learn more about the SDK build system.

fanghuaqi, 2020-12-01:
> 
> 
> @fanghuaqi Ok. Do you want me to take shot at this? Would be a good way for me to learn more about the SDK build system.

OK, you can create your feature branch and implement this feature, I can review it.

fanghuaqi, 2020-12-01:
> 
> 
> Is it also possible to set the SOC to gd32vf103 by default for the longan nano board?

If you want to set the default board to longan nano, there is a way to do it without change any code base.

See https://doc.nucleisys.com/nuclei_sdk/develop/appdev.html#set-default-make-options

RomanBuchert, 2020-12-01:
> If you want to set the default board to longan nano, there is a way to do it without change any code base.

@fanghuaqi  I want to do it the other way. If one of the gd32vf103 boards are selected, than the SOC should automaticly be set to gd32vf103.

fanghuaqi, 2020-12-01:
> 
> 
> > If you want to set the default board to longan nano, there is a way to do it without change any code base.
> 
> @fanghuaqi I want to do it the other way. If one of the gd32vf103 boards are selected, than the SOC should automaticly be set to gd32vf103.

Currently it is not possible, `BOARD` variable is defined under `SOC`, so the `SOC` variable is a must, and also the `BOARD` name could be the same for different `SOC`s

fanghuaqi, 2020-12-03:
Fixed by #18

closed 2020-12-03 by fanghuaqi