Nuclei-Software/nuclei-sdk #24: VSCode Plugin for Nuclei SDK

fanghuaqi, 2020-12-17:
Maybe we can have a VSCode plugin for nuclei sdk in future, which provide the following features:

* Automatically download or use existing Nuclei SDK source code base
* Create or use existing nuclei-sdk project as templates
* Create project wizard support choosen of SoC, Board, Core, Download
* Support build and debug application directly in VSCode using Makefile build system

Currently we have `platformio-nuclei` which also provide `Nuclei SDK` intergation in VSCode, but maybe there will be a good try to have a native plugin to directly use Nuclei SDK build system.

jardinimf, 2021-03-14:
Microsoft now have a preview tool for Makefile projects (microsoft/vscode-makefile-tools here at github).

I was able to use nuclei-sdk on VScode without need of setup_config(.sh/.bat), only needed changes are in "$(NUCLEI_SDK_ROOT)/Build/Makefile.conf" line 7:

from: NUCLEI_RISCV_GCC_ROOT ?= $(NUCLEI_SDK_TOOL_ROOT)/$(COMPILE_PREFIX)gcc
to: NUCLEI_RISCV_GCC_ROOT ?= $(NUCLEI_SDK_TOOL_ROOT)/gcc

After that, add the path to Nuclei's gcc as "NUCLEI_SDK_TOOL_ROOT=..." instead of "NUCLEI_TOOL_ROOT=..." in the program Makefile, or in Makefile.local (together with program Makefile) or as Makefile.global in "nuclei-sdk/Build"

hint: the default search path for NUCLEI_SDK_TOOL_ROOT is $(NUCLEI_SDK_ROOT)/prebuilt_tools

I'm following the recommended dir strutct of "Nuclei/gcc" and "Nuclei/openocd" for nuclei tools, like in the README.md

* this change also works for building outside VScode

fanghuaqi, 2021-03-15:
Thank you @jardinimf for your detailed investigation, I will accept you suggestion and update the Makefile later.

fanghuaqi, 2026-11-12:
Nuclei Studio VSCode Plugin has been released, see https://doc.nucleisys.com/nuclei_tools/vscode/install.html , it can be used with Nuclei Studio 2026.10 release.

closed 2026-11-12 by fanghuaqi