Nuclei-Software/nuclei-sdk #31: Error in __set_exc_entry

DagAgren, 2022-02-18:
The comments for __set_exc_entry says that address should be aligned to 4 bytes, but the code itself forces alignment to 64 bytes:

https://github.com/Nuclei-Software/nuclei-sdk/blob/master/NMSIS/Core/Include/core_feature_eclic.h#L757

This looks like a mistake copied from the earlier n200_func.c file, which also incorrectly aligned the address to 64 bytes. The address should be masked by `~3`, not `~0x3f`.

This code probably works by accident since a lot of startup code seems to place the interrupt table at the start of Flash memory, so it is accidentally aligned.

fanghuaqi, 2022-02-21:
Thanks for your reporting, I will take it into bug fix list.

fanghuaqi, 2022-02-24:
Fixed the comment

closed 2022-02-28 by fanghuaqi