Nuclei-Software/nuclei-sdk #35: arithmetic on `void*`

wallacegibbon, 2022-03-16:
In `NMSIS/Core/Include/core_feature_timer.h`, there are codes like:
```c
addr = (void *)(&(SysTimer->MTIMER));
high0 = __LW(addr + 4);
```
which will cause many compiling warnings in g++.

When doing arithmetic operations, `void *` will be treated as `char *` in GCC, but it is not defined in the C standard,
I guess it will be better to replace them by something like

```c
addr = (void *)(&(SysTimer->MTIMER));
high0 = __LW((int *) addr + 1);
```

or 

```c
addr = (void *)(&(SysTimer->MTIMER));
high0 = __LW((char *) addr + 4);
```

fanghuaqi, 2022-03-17:
> which will cause many compiling warnings in g++.

Could you show me what warning is throw out by g++? thanks

wallacegibbon, 2022-03-17:
OK, here are the warnings copied from my terminal:

```
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h: In function 'void SysTimer_SetLoadValue(uint64_t)':
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h:123:15: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
  123 |     __SW(addr + 4, (uint32_t)(value >> 32));
      |          ~~~~~^~~
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h: In function 'uint64_t SysTimer_GetLoadValue()':
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h:148:23: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
  148 |     high0 = __LW(addr + 4);
      |                  ~~~~~^~~
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h:150:22: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
  150 |     high = __LW(addr + 4);
      |                 ~~~~~^~~
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h: In function 'void SysTimer_SetCompareValue(uint64_t)':
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h:180:19: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
  180 |         __SW(addr + 4, (uint32_t)(value >> 32));
      |              ~~~~~^~~
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h:189:19: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
  189 |         __SW(addr + 4, (uint32_t)(value >> 32));
      |              ~~~~~^~~
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h: In function 'uint64_t SysTimer_GetCompareValue()':
./nuclei-sdk/NMSIS/Core/Include/core_feature_timer.h:218:26: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
  218 |         high = __LW(addr + 4);
      |                     ~~~~~^~~

```

fanghuaqi, 2022-03-31:
Fixed

closed 2022-03-31 by fanghuaqi