Nuclei-Software/nuclei-sdk #39: Prebuilt libraries disk usage

gsauthof, 2022-05-01:
When cloning the repository over a slow connection I noticed that it contains a bunch of prebuilt libraries:

```
$ du -chs NMSIS/Library/
117M	NMSIS/Library/
117M	total
```

I'm kind of afraid that those prebuilt libraries will bloat this repository in the future even more, when each change to them basically adds another 100 MBs or so to the git index.

Is it really necessary to include those artifacts?

Perhaps there are alternatives such as using Git Large File Storage or building them from source like the rest?

fanghuaqi, 2022-05-05:
Hi @gsauthof , currently this prebuilt libraries are used by demo_dsp example, and with this prebuilt NMSIS DSP/NN libraries at specified version of NMSIS, user no need to build this library.

If I add this as Git LFS objects, it might introduce more complicated usage for some basic user which is not familiar with git, and maybe in future I can add it as git lfs, and provide a better solution when I create regular releases, and normal user should use the regular release version of SDK, and advance user directly use with git version.

Thanks
Huaqi

fanghuaqi, 2023-11-07:
With the release 0.5.0, more variant of prebuilt libraries are included, so the size of library now is 425M.

~~~
$ du -d1 -h NMSIS/Library/
361M    NMSIS/Library/DSP
64M     NMSIS/Library/NN
425M    NMSIS/Library/
~·~~

fanghuaqi, 2024-11-11:
In future release of Nuclei SDK, we will directly include NMSIS DSP/NN source code, instead of library, so user can directly try with the source code, not prebuilt library with big size, and can build with different arch