Nuclei-Software/nuclei-sdk #45: SMP demo fail, what can i do?

zhangyang-hxd, 2022-06-17:
I create a smp demo by "SMP Helloworld in baremetal environment @app-nsdk_smphello" on NucleiDemo.
But it can't run into main() when i debug it with qemu, does i need to do any other configuration?
I also build it from nuclei-sdk/application/baremetal/smphello by hand. Only one cpu run.
LOGS:
"Nuclei SDK Build Time: Jun 15 2022, 18:01:00
Download Mode: ILM
CPU Frequency 46858 Hz
Hello world from hart 0
Some harts boot failed, only 1/2 booted!"
I think nuclei-n307fd which demo use by default does not support smp?

I test by below:
=========================
qemu-system-riscv32.exe  -machine nuclei_n -cpu nuclei-n307fd -smp 2 -serial stdio -nodefaults -S -s
qemu-system-riscv32.exe: Invalid SMP CPUs 2. The max CPUs supported by machine 'nuclei_n' is 1
=========================================

fanghuaqi, 2022-06-20:
Qemu not supported in 2022.04 release version, we will add this smp feature in qemu in future releases.

zhangyang-hxd, 2022-06-20:
Ok, thanks。
So， if i want to test smp feature, i need to verify it on SOC, right? Which SOC i can use, UX900?

fanghuaqi, 2022-06-20:
Yes, you need to contact our sales or AE to for support on UX900-MC, if you have our UX900-MC core, you can test it on fpga.

Check link here for how to run: https://doc.nucleisys.com/nuclei_sdk/design/app.html#smphello

zhangyang-hxd, 2022-06-20:
ok，thank you very much for your reply。
I will try above。

closed 2022-06-20 by fanghuaqi