Nuclei-Software/nuclei-sdk #46: Sipeed-RV-debugger-lite supported?

KafCoppelia, 2022-07-16:
I followed [nuclei_sdk Manual](https://doc.nucleisys.com/nuclei_sdk/quickstart.html) on Linux with longan nano GD3VF103CBT6.Building example `./application/baremetal/helloworld` according to the manual is OK. Then I uploaded with sipeed-rv-debugger-lite by `make SOC=gd32vf103 BOARD=gd32vf103c_longan_nano upload`:
```
Download and run helloworld.elf
riscv-nuclei-elf-gdb helloworld.elf -ex "set remotetimeout 240" \
-ex "target remote | openocd  -c \"; gdb_port pipe; log_output openocd.log\" -f ../../../SoC/gd32vf103/Board/gd32vf103c_longan_nano/openocd_gd32vf103.cfg" \
--batch -ex "thread apply all monitor reset halt" -ex "thread apply all set \$pc=_start" -ex "thread 1" -ex "load" -ex "monitor resume" -ex "quit"
Open On-Chip Debugger 0.11.0+dev-01897-gc8fc2a168 (2022-04-01-08:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
_start0800 () at ../../../SoC/gd32vf103/Common/Source/GCC/startup_gd32vf103.S:372
372         j 1b

Thread 1 (Remote target):
JTAG tap: riscv.cpu tap/device found: 0x1000563d (mfg: 0x31e (Andes Technology Corporation), part: 0x0005, ver: 0x1)
JTAG tap: auto0.tap tap/device found: 0x790007a3 (mfg: 0x3d1 (GigaDevice Semiconductor (Beijing) Inc), part: 0x9000, ver: 0x7)

Thread 1 (Remote target):
[Switching to thread 1 (Remote target)]
#0  _start () at ../../../SoC/gd32vf103/Common/Source/GCC/startup_gd32vf103.S:214
214         csrc CSR_MSTATUS, MSTATUS_MIE
Loading section .init, size 0x2f4 lma 0x8000000
Loading section .text, size 0x254c lma 0x8000300
Loading section .rodata, size 0x490 lma 0x800284c
Loading section .data, size 0x6c lma 0x8002cdc
Start address 0x08000180, load size 11580
Transfer rate: 2 KB/sec, 2895 bytes/write.
A debugging session is active.

        Inferior 1 [Remote target] will be detached.

Quit anyway? (y or n) [answered Y; input not from terminal]
[Inferior 1 (Remote target) detached]
```
However, the serial output is wired (115200 8N1):
```
N], *
iTZ  &2,0:M
oXjde'21A*
A

AU uYk L47LB

!* b MR 090M
SR *IS
UXCH0:BllrFN], j5Vooj
 HY Kd2mrlYZ%MCKocYrH
&'BlWKd m leZ%ScYK
CHeWK  eZ%Sooj
!HYWKd2omrclY%SCKocY!CCBlorl2mrlYZ%MCKcYrHCH lol2o[rcRRcYK
CHYWK  eZ%Sooj
! HY Kd2mrlYZ%SCKocY!CC lorl2o[rc RRVcYrHCH(:BlrFN], j5VocY!CC:Blll2o[rclY%Sooj
: lorl2o[rcRRVcYrHCH1&HY Kd2mrlYZ%SCKocY!CC:Blll2omrclY%Sooj
: lorl2o[rcRRVcYK
C HY Kd2mrlYZ%SCKocYrH
1HYWK  eZ%Sooj
: lorl2o[rcRRVcYK
C HY Kd2mrlYZ%MCKocYrH
```
I tried different baudrate(9600...) but none of it was right. 

Besides, multiple reset will get different results.

fanghuaqi, 2022-07-18:
Is the uart port used by is TX0/RX0 below
[image]

fanghuaqi, 2022-07-18:
I didn't try `Sipeed-RV-debugger-lite` debugger before, so I can promise it will work. The COM port used is https://github.com/Nuclei-Software/nuclei-sdk/blob/master/SoC/gd32vf103/Board/gd32vf103c_longan_nano/Include/nuclei_sdk_hal.h#L11

KafCoppelia, 2022-07-18:
> Is the uart port used by is TX0/RX0 below
> [image]
> 

Of course it is TX0/RX used in debugging.

fanghuaqi, 2022-07-18:
Baudrate is 115200bps，you can try to connect the rx/tx port using other serial hardware.

KafCoppelia, 2022-07-22:
使用rv-debugger lite下载，但是换了tx rx连接至另外一个串口转usb硬件，下载顺利，但是，115200输出如下：
```
N], *
iTZ 20&b L1'
C!l+ S:21A*
A
CAU2e\Wc^01N J5
!*t,j%SP'x0L5CHMR *IS
U
! lrFN], j5Vooj
 lrFN], j5Vooj
! lrFN], j5Vooj
 lrFN], j5Vooj
!HY Kd2o[rc j5Vooj
HY Kd2o[rc j5Vooj
!HY Kd2o[rc j5Vooj
HY Kd2o[rc j5Vooj
! lrFN], j5Vooj
 lrFN], j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
:BlWKd2o[rc j5Vooj
```
状况稍微好了一点

KafCoppelia, 2022-07-24:
今日对我串口工具软件的设置进行了修改，当我不使用rv-debugger-lite上的串口时（TX/RX另外接到其他的USB转串口硬件），发现串口可以正常读取数据了，baudrate确实为115200。但是若用rv-debugger-lite上的串口时，仍然不对。

fanghuaqi, 2022-07-25:
> 今日对我串口工具软件的设置进行了修改，当我不使用rv-debugger-lite上的串口时（TX/RX另外接到其他的USB转串口硬件），发现串口可以正常读取数据了，baudrate确实为115200。但是若用rv-debugger-lite上的串口时，仍然不对。

嗯嗯，那说明起码程序是OK的，只是rv-debugger-lite上的转出来的串口可能哪里不对，这个可能就需要反馈给厂家那边了。

KafCoppelia, 2022-07-25:
rv-debugger-lite在我linux上，用`lsusb`查看：
```
Bus 003 Device 004: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
```

fanghuaqi, 2022-07-25:
> rv-debugger-lite在我linux上，用`lsusb`查看：
> 
> ```
> Bus 003 Device 004: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC
> ```
> 
> 不知道是不是这块芯片的问题

你看看这个调试器上的芯片是不是这个，我这边不是这个调试器的厂家，具体的实现，可能得咨询你那边的卖家

KafCoppelia, 2022-07-25:
是CH552T，这不是FT2232

fanghuaqi, 2022-07-26:
我猜那可能是模拟实现的ft2232。这样子这个问题就先关闭了，应该是需要将 longan板子上的tx/rx单独接到另外的usb转串口的模块上就可以正常的收发串口了。

According to discussion above in Chinese, the TX/RX pins in longan board need to be connected to another USB-TTL serial hardware, and then the UART can receive correct characters using 115200bps, and there might  be some issue with the serial on the longan debugger hardware.

This issue closed now.

closed 2022-07-26 by fanghuaqi

KafCoppelia, 2022-07-26:
好的，这个问题我看pio论坛上也有类似用sipeed rv-debugger-lite翻车的，pio对这个调试器仿佛也不支持。即使他们通过 一些修改可以使用lite下载，串口波特率也不对。类似issue：[Support Sipeed RV-Debugger Lite debug probe](https://github.com/sipeed/platform-gd32v/issues/31)、[Sipeed RV-Debugger Lite](https://community.platformio.org/t/sipeed-rv-debugger-lite/19784/2)

fanghuaqi, 2022-07-28:
目前来看根据这里的描述 https://community.platformio.org/t/sipeed-rv-debugger-lite/19784/14 貌似其他的波特率可以工作，你可以做下实验 修改波特率 https://github.com/Nuclei-Software/nuclei-sdk/blob/master/SoC/gd32vf103/Board/gd32vf103c_longan_nano/Source/gd32vf103c_longan_nano.c#L123。

> Another interesting feature-ette of the Sipeed RV-Debugger Lite is that the serial port on it does not support 115200.
> 
> Just the following:-
> 2400 4800 9600 19200 38400 43000 56000 57600 128k 256k
> 
> I guess 115200 must be a division of the micro-controllers clock that is not supported.
> 
> However the next fastest speed of 128000 seems to work OK
> 
> --- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
> --- More details at http://bit.ly/pio-monitor-filters
> --- Miniterm on /dev/ttyUSB1  128000,8,N,1 ---
> --- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> 
> USART0 Initialised to 128000 8N1 no flow control
> 
> Booted
> Device Info:-
> Flash:      64kB
> SRAM:       20kB
> Device ID: 3934057C 00053935 FFFFFFFF
> Having a debugger that also has a serial port to monitor output during the development phase without needing to plug in a separate device is handy. Just watch out for what is for many (particularly those used to AVR and Arduino) the default baud rate not being supported.

KafCoppelia, 2022-07-28:
原来sdk这里也要改，那看来是改成128000成功的概率很大。不幸的是，我今天重新编译的时候遇到了共享库缺失的问题，但我不知道原因，我也重新安装了nuclei-gcc，依然如此。当我`upload`时，报错：
```
Download and run helloworld.elf
riscv-nuclei-elf-gdb helloworld.elf -ex "set remotetimeout 240" \
-ex "target remote | openocd  -c \"; gdb_port pipe; log_output openocd.log\" -f ../../../SoC/gd32vf103/Board/gd32vf103c_longan_nano/openocd_gd32vf103.cfg" \
--batch -ex "thread apply all monitor reset halt" -ex "thread apply all set \$pc=_start" -ex "thread 1" -ex "load" -ex "monitor resume" -ex "quit"
riscv-nuclei-elf-gdb: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory
make: *** [../../../Build/Makefile.rules:124：upload] 错误 127
```

KafCoppelia, 2022-07-28:
库缺失的问题解决了，在linux上解决远比win上麻烦，可能是我在升级其他包的时候破坏了这个依赖。测了128000，用cutecom可以很好的显示：
```
00000000 4e 75 63 6c 65 69 20 53   44 4b 20 42 75 69 6c 64 	Nuclei S  DK Build
00000016 20 54 69 6d 65 3a 20 4a   75 6c 20 32 38 20 32 30 	 Time: J  ul 28 20
00000032 32 32 2c 20 32 31 3a 33   38 3a 34 36 0d 0d 0a 44 	22, 21:3  8:46␍␍␊D
00000048 6f 77 6e 6c 6f 61 64 20   4d 6f 64 65 3a 20 46 4c 	ownload   Mode: FL
00000064 41 53 48 58 49 50 0d 0d   0a 43 50 55 20 46 72 65 	ASHXIP␍␍  ␊CPU Fre
00000080 71 75 65 6e 63 79 20 31   30 37 34 38 35 37 31 34 	quency 1  07485714
00000096 20 48 7a 0d 0d 0a 48 61   72 74 20 30 2c 20 4d 49 	 Hz␍␍␊Ha  rt 0, MI
00000112 53 41 3a 20 30 78 34 30   39 30 31 31 30 35 0d 0d 	SA: 0x40  901105␍␍
00000128 0a 4d 49 53 41 3a 20 52   56 33 32 49 4d 41 43 55 	␊MISA: R  V32IMACU
00000144 58 0d 0d 0a 30 3a 20 48   65 6c 6c 6f 20 57 6f 72 	X␍␍␊0: H  ello Wor
00000160 6c 64 20 46 72 6f 6d 20   4e 75 63 6c 65 69 20 52 	ld From   Nuclei R
00000176 49 53 43 2d 56 20 50 72   6f 63 65 73 73 6f 72 21 	ISC-V Pr  ocessor!
00000192 0d 0d 0a 31 3a 20 48 65   6c 6c 6f 20 57 6f 72 6c 	␍␍␊1: He  llo Worl
00000208 64 20 46 72 6f 6d 20 4e   75 63 6c 65 69 20 52 49 	d From N  uclei RI
00000224 53 43 2d 56 20 50 72 6f   63 65 73 73 6f 72 21 0d 	SC-V Pro  cessor!␍
00000240 0d 0a 32 3a 20 48 65 6c   6c 6f 20 57 6f 72 6c 64 	␍␊2: Hel  lo World
00000256 20 46 72 6f 6d 20 4e 75   63 6c 65 69 20 52 49 53 	 From Nu  clei RIS
00000272 43 2d 56 20 50 72 6f 63   65 73 73 6f 72 21 0d 0d 	C-V Proc  essor!␍␍
00000288 0a 33 3a 20 48 65 6c 6c   6f 20 57 6f 72 6c 64 20 	␊3: Hell  o World 
00000304 46 72 6f 6d 20 4e 75 63   6c 65 69 20 52 49 53 43 	From Nuc  lei RISC
00000320 2d 56 20 50 72 6f 63 65   73 73 6f 72 21 0d 0d 0a 	-V Proce  ssor!␍␍␊
00000336 34 3a 20 48 65 6c 6c 6f   20 57 6f 72 6c 64 20 46 	4: Hello   World F
00000352 72 6f 6d 20 4e 75 63 6c   65 69 20 52 49 53 43 2d 	rom Nucl  ei RISC-
00000368 56 20 50 72 6f 63 65 73   73 6f 72 21 0d 0d 0a 35 	V Proces  sor!␍␍␊5
00000384 3a 20 48 65 6c 6c 6f 20   57 6f 72 6c 64 20 46 72 	: Hello   World Fr
00000400 6f 6d 20 4e 75 63 6c 65   69 20 52 49 53 43 2d 56 	om Nucle  i RISC-V
00000416 20 50 72 6f 63 65 73 73   6f 72 21 0d 0d 0a 36 3a 	 Process  or!␍␍␊6:
00000432 20 48 65 6c 6c 6f 20 57   6f 72 6c 64 20 46 72 6f 	 Hello W  orld Fro
00000448 6d 20 4e 75 63 6c 65 69   20 52 49 53 43 2d 56 20 	m Nuclei   RISC-V 
00000464 50 72 6f 63 65 73 73 6f   72 21 0d 0d 0a 37 3a 20 	Processo  r!␍␍␊7: 
00000480 48 65 6c 6c 6f 20 57 6f   72 6c 64 20 46 72 6f 6d 	Hello Wo  rld From
00000496 20 4e 75 63 6c 65 69 20   52 49 53 43 2d 56 20 50 	 Nuclei   RISC-V P
00000512 72 6f 63 65 73 73 6f 72   21 0d 0d 0a 38 3a 20 48 	rocessor  !␍␍␊8: H
00000528 65 6c 6c 6f 20 57 6f 72   6c 64 20 46 72 6f 6d 20 	ello Wor  ld From 
00000544 4e 75 63 6c 65 69 20 52   49 53 43 2d 56 20 50 72 	Nuclei R  ISC-V Pr
00000560 6f 63 65 73 73 6f 72 21   0d 0d 0a 39 3a 20 48 65 	ocessor!  ␍␍␊9: He
00000576 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000592 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000608 63 65 73 73 6f 72 21 0d   0d 0a 31 30 3a 20 48 65 	cessor!␍  ␍␊10: He
00000624 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000640 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000656 63 65 73 73 6f 72 21 0d   0d 0a 31 31 3a 20 48 65 	cessor!␍  ␍␊11: He
00000672 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000688 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000704 63 65 73 73 6f 72 21 0d   0d 0a 31 32 3a 20 48 65 	cessor!␍  ␍␊12: He
00000720 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000736 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000752 63 65 73 73 6f 72 21 0d   0d 0a 31 33 3a 20 48 65 	cessor!␍  ␍␊13: He
00000768 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000784 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000800 63 65 73 73 6f 72 21 0d   0d 0a 31 34 3a 20 48 65 	cessor!␍  ␍␊14: He
00000816 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000832 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000848 63 65 73 73 6f 72 21 0d   0d 0a 31 35 3a 20 48 65 	cessor!␍  ␍␊15: He
00000864 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000880 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000896 63 65 73 73 6f 72 21 0d   0d 0a 31 36 3a 20 48 65 	cessor!␍  ␍␊16: He
00000912 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000928 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000944 63 65 73 73 6f 72 21 0d   0d 0a 31 37 3a 20 48 65 	cessor!␍  ␍␊17: He
00000960 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00000976 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00000992 63 65 73 73 6f 72 21 0d   0d 0a 31 38 3a 20 48 65 	cessor!␍  ␍␊18: He
00001008 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00001024 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00001040 63 65 73 73 6f 72 21 0d   0d 0a 31 39 3a 20 48 65 	cessor!␍  ␍␊19: He
00001056 6c 6c 6f 20 57 6f 72 6c   64 20 46 72 6f 6d 20 4e 	llo Worl  d From N
00001072 75 63 6c 65 69 20 52 49   53 43 2d 56 20 50 72 6f 	uclei RI  SC-V Pro
00001088 63 65 73 73 6f 72 21 0d   0d 0a                   	cessor!␍  ␍␊
```
但是用另一个串口工具moserial，同样的串口设置，却显示乱码，可能是软件不同导致的。