Nuclei-Software/nuclei-sdk #47: Register keyword removal

gsauthof, 2022-09-22:
I noticed that some headers use the `register` keyword, e.g.:

https://github.com/Nuclei-Software/nuclei-sdk/blob/b188abafa42e3a4405c298f97ad5b740cbe8215e/NMSIS/Core/Include/core_feature_base.h#L1487-L1502

AFAICS, those `register` keywords are superfluous because the following inline assembler directives specify the register-constraint for those variables.

However, OTOH, the asm register constraint can't be left out because the C register keyword is just a hint (in contrast to the [GCC local register variable extension](https://gcc.gnu.org/onlinedocs/gcc/Local-Register-Variables.html#Local-Register-Variables) which is only supposed to be used if constraints aren't sufficient).

So I suggest to simply remove those `register` keywords.

---

Background: The register keyword was deprecated in C++ 17 and hence GCC warns about this when compiling with C++17 features enabled, even when wrapping the `gd32vf103` headers in `extern "C" { }`.

As a workaround I `#define register` before including and gd32vf103 headers and so far I don't see any issues.

At least for now, one can also compile with `-Wno-register`.

fanghuaqi, 2022-09-23:
Hi @gsauthof , thank you for the suggestion, I will take a try with removing it, see whether affect other source code.

fanghuaqi, 2022-09-26:
See latest develop branch, register keywords are removed

closed 2022-09-26 by fanghuaqi