Nuclei-Software/nuclei-sdk #48: GD32VF103: Update driver?

Headcrabed, 2022-10-08:
Currently, SDK is still using the driver which is 2 years old. Driver from gigadevice has updated a lot since then, so better update to newer version.

fanghuaqi, 2022-10-09:
Any volunteer to handle this? If you are interested, or you have some internal stuffs, you can try to upgrade it and make a PR, I can help to review it.

This is the latest firmware library website https://www.gd32mcu.com/en/download/7?kw=GD32VF1

> latest version at 2022.10.9 is 1.1.4

[image]

Headcrabed, 2022-10-09:
Seems some modification on SDK is needed. I tried directly changing firmware library to newest version but it reports tons of errors.

fanghuaqi, 2022-10-09:
> Seems some modification on SDK is needed. I tried directly changing firmware library to newest version but it reports tons of errors.

You need to carefully upgrade the firmware, some changes might need to be done in Makefile of gd32vf103, you can take a try in your side, and tried to make it work at least all the applications can successfully run in your board, then the upgrading is somehow ready for PR review.

closed 2023-10-18 by fanghuaqi