Nuclei-Software/nuclei-sdk #49: How to use Clang/llvm compile the SDK?

sunch20, 2022-12-07:
The SDK uses riscv-nuclei-elf-gcc to compile by default. If I want to compile with Clang, what needs to be modified for the Makefile?

fanghuaqi, 2022-12-07:
Hi @sunch20 , currently only gcc is supported, if you need to support other toolchain, you need to modify `Build` folder, and `SoC/xxx/build.mk`, and maybe linker script file for each soc is also need to be updated.

sunch20, 2022-12-07:
> Hi @sunch20 , currently only gcc is supported, if you need to support other toolchain, you need to modify `Build` folder, and `SoC/xxx/build.mk`, and maybe linker script file for each soc is also need to be updated.
Okay, thanks for your advice. I have one more question though.Besides using riscv-nuclei-elf-gcc to compile the SDK, can I use riscv-none-embed-gcc to compile？

fanghuaqi, 2023-09-19:
If you want to take a try with llvm toolchain now, you can try this #53

closed 2023-10-18 by fanghuaqi