Nuclei-Software/nuclei-sdk #59: Is a bug in CTEST?

yinqingw, 2024-04-01:
in `test_csr.c`:
```
CTEST(core, icache)
{
    EnableICache();
    ASSERT_NOT_EQUAL(__RV_CSR_READ(CSR_MCACHE_CTL) & CSR_MCACHE_CTL_IE, 0);
    DisableICache();
    ASSERT_EQUAL(__RV_CSR_READ(CSR_MCACHE_CTL) & CSR_MCACHE_CTL_IE, 0);
}
```
should do 
```
__RWMB();
__FENCE_I();
```
before `EnableICache();`and`EnableICache();`?
Now，If my compilation option is `-02`,CPU may be abnormal。

fanghuaqi, 2024-04-02:
Sorry, I cannot repeat it, could you provide more details for what cpu you are using, and changes you have made?

Tested on nx900fd and n900fd with `-O2`.

[image]

closed 2024-04-02 by fanghuaqi

yinqingw, 2024-04-02:
It is a possible trouble，you could only run 
```
EnableICache();
DisableICache();
EnableICache();
DisableICache();
EnableICache();
DisableICache();
```
for enable and disable icache。It  may have appeared.

yinqingw, 2024-04-03:
ok,thanks