Nuclei-Software/nuclei-sdk #61: Using Nuclei C Runtime library, malloc failed？

yinqingw, 2024-12-20:
In my project，STDCLIB ?= libncrt_balanced；__HEAP_SIZE=2K;
But, void *ptr = malloc(100);
ptr = 0???
Why did this situation occur？
When STDCLIB ?= newlib_full, it is ok.

fanghuaqi, 2024-12-23:
Which branch you are using, give me a sample app to test it.

fanghuaqi, 2024-12-23:
If you are doing your customized SoC, you need to follow the linker script template based on evalsoc, and stub files for libncrt, did you check it?

Close it since not repeatable.

closed 2024-12-23 by fanghuaqi

yinqingw, 2024-12-23:
ok, I find miss `call __libc_init_array` in startup.S. Fixed it, it work ok. Thanks!

fanghuaqi, 2024-12-23:
Yes, it is required for setup libncrt heap, see https://github.com/Nuclei-Software/nuclei-sdk/blob/670dfbe535b35a0411bb72902937a20a9864ca3d/SoC/evalsoc/Common/Source/Stubs/libncrt/stubs.c#L54-L58